dvhdr/launchpad-pro #5: Anyway to hack on this only when in User mode?

malkomalko, 2015-07-24:
I love what you are doing with this and am excited to start hacking on this, but was wondering if there was anyway to get the benefits of the default firmware, while only hacking if you are in user mode.  I know the "Launchpad Pro-1.0.154.syx" is shipped, but this seems to be an all or nothing approach yes?

Cheers

dvhdr, 2015-07-24:
Yes, all or nothing for now.  As described in the README, our philosophy was to remove almost all of the factory functionality for a variety of reasons. Once we learn from how this first "bare bones" release goes, we'll certainly consider other possibilities.  Thanks for the feedback!

dvhdr, 2015-07-24:
...re-reading the README, it's not as clear as it could be.  I'll update it.

faroit, 2015-07-30:
I think many people would like to add just a few features in the keyboard or drum mode. 

E.g. I would love to have a side button functioning as sustain. This would be fairly easy to implement if I do not have to implement the keyboard functionality myself. Therefore I think for many people it would very beneficial to make those modes open source, too. 

I guess this would be a very tough business decision for novation but me feeling is that I some of the development for the launchpad pro firmware would take place here, people would love novation even more resulting in that you guys sell more hardware thus making everyone more happy ;-)

dvhdr, 2015-07-31:
Thanks loads for your feedback!  I expect you're not alone in wanting to lightly customise the unit, instead of build completely new apps for it.  Much food for thought here... though I can't make any promises at this point, other than to seriously consider the options!

hotchk155, 2015-10-02:
Would it be possible to port certain factory firmware features like drum mode and keyboard mode into your API framework, so that you could release the source code for those parts only, without exposing any of the low level magic?

JacobiusWrex, 2016-07-02:
I am an electrician with experience programming PLCs in assembly lines, but my experience is only through guis in function block or ladder logic type languages. I am very motivated to develop some cool custom interaction with Propellerhead Reason. I dont even know how to start with this though.. Could someone point me in the right direction for learning how to work in this language?

dvhdr, 2018-11-08:
Check out @mat1jaczyyy's work on https://github.com/mat1jaczyyy/lpp-performance-cfw

closed 2018-11-08 by dvhdr