dvhdr/launchpad-pro #9: LPPro with sample app.c not enumerating as MIDI device (Win7x64)

hotchk155, 2015-09-18:
Hi - very excited about using this API and got an LPPro after seeing this project. I've been able to run the sample app after building and uploading via SysEx however my LPPro does not show up as a MIDI device under Win7x64 when sample app is running. It is just enumerating as a generic "USB Composite Device" with a "Cannot Start" error when the sample app is installed. It seems to be fine after restoring factory firmware.

Reading the documentation I think I should see it show up as a MIDI device which should be able to receive input from the sample app. I also understand no drivers are needed

Is my understanding incorrect regarding how this is supposed to work?

Really looking forward to getting this working.. I've been hoping for something like this for ages!

Cheers
Jason

dvhdr, 2015-09-18:
Ick, you're right, I see the same issue. I'm just glad you've got another machine (OS X I guess) that you can restore the factory firmware from!  I guess I must have made a mistake with the descriptor when I changed it.  I'll investigate as soon as I can - thanks for the report.

hotchk155, 2015-09-18:
Thanks Dave, at least I know its not just me! By the way the LP still enumerates fine in bootloader mode so I can push the factory firmware back to it without too much of a problem

dvhdr, 2015-09-18:
Yes.  I forgot for a second as nightmare visions of people stuck with bricked units clouded my thinking!

Going to delve into the descriptors then...

hotchk155, 2015-10-02:
Just wondering if there is any news on the descriptor issue..? Thanks/Jason

dvhdr, 2015-10-02:
I took a quick look, hoping I'd broken something trivial. However, I didn't find anything dumb, so I'm now wondering if my port to GCC-ARM-Eclipse changed something more fundamental. So it's time to get the USB analyser out and delve more deeply, which is a major gumption trap.

I do hope to get onto it in the next few weeks though, I love the look of your sequencer project!

The-XOR, 2015-11-24:
Same issue here (Win10x64). Any news/possible date for a fix?
Thank you.

Franvu, 2015-12-26:
any news?
Cheers.

RixdeTurcan, 2016-05-29:
Hi,

Is there any news about the fix of this bug ?

I am using windows 7 64 bit, and I have the same issue.

I can see also that the plug-and-play service is running anormally (the service does not work, and use the cpu erraticaly with many peaks) after trying to connect the Lauchpad pro with a flashed openfirmware (the computer cannot be stopped normally, i have to power it off).

Is there any possibility to turn around this bug ? I bought a launchpad in order to program it for my applications, and i'm sad to see that it isn't possible.

thank you,
Jory

Franvu, 2016-05-29:
No news. I have the same problem and I wanted to program my applications too.
Finally,  I use this program to play with launchpad:
http://fabriziopoce.com/isocontrollers.html
The developer is very friendly.
I wish there was an official program like this.
Cheers.

The-XOR, 2016-05-29:
Ages ago,  I contacted Novation support signaling this iasue,  that would take a couple of minutes to be fixed.  They are not interested in supporting their own devices.

dvhdr, 2016-05-30:
Hi all. I would fix it if I could isolate the cause! It appears there's something subtly wrong (nothing glaringly obvious). Next steps (which I haven't yet taken) are to:

-run the Chapter 9 tests and see if there's any clues
-regress to earlier versions of the toolchain / build settings / optimisations

Sorry, but I can't make any promises about when I might get a chance to do this.

RixdeTurcan, 2016-05-31:
Franvu -> I found a workaround which is working for me (and should work for anyone having this issue) : 
- Install the free virtual midi loop (it send what it get) device "LoopBe1".
- Set the launchpad in programmation mode.
- Write a bunch of python code to interface the Launchpad with the virtual midi device (using "rtmidi_python").
- In your favorite music software, connect it to the virtual midi device instead of the launchpad.

If you have only applications with a pc, I think that this solution is better than program the launchpad (no cpu saturation of the µcontroller, you can code with floats and all of your favorite libraries, you can do fun things, like launch musics on the pc or control the mouse pointer/keyboard using the launchpad).

But if you want to interface the launchpad with another midi hardware, you have to reprogram it, and you will not be able to connect it to a pc.. but I think it is better than waiting for a 8 month-aged bugfix..

SeanM93, 2016-06-14:
I've been trying to find a workaround that involves editing the .inf file from the standard Novation USB MIDI driver and forcing it to load. Unfortunately I have no idea what I am doing, but hopefully this prompts somebody who can come up with a fix. 

Going through the Windows device manager I found in the error messages for the USB Composite Device (i.e. the Launchpad Pro I have flashed the example to) I got the ID for the device. 

`USB\Vid_1235&Pid_4051\Launchpad_Open`

Inside the .inf file, I found 

```
%LaunchpadPro1.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_0051
%LaunchpadPro2.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_0052
%LaunchpadPro3.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_0053
%LaunchpadPro4.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_0054
%LaunchpadPro5.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_0055
%LaunchpadPro6.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_0056
%LaunchpadPro7.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_0057
%LaunchpadPro8.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_0058
%LaunchpadPro9.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_0059
%LaunchpadPro10.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_005a
%LaunchpadPro11.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_005b
%LaunchpadPro12.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_005c
%LaunchpadPro13.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_005d
%LaunchpadPro14.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_005e
%LaunchpadPro15.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_005f
%LaunchpadPro16.DeviceDesc%=	DDInstall,USB\Vid_1235&Pid_0060
```

This appears twice, once under models and again under the 64 bit models (i presume to direct it to the 64 bit driver). 

I tried adding `%LaunchpadProOpen1.DeviceDesc%= DDInstall,USB\Vid_1235&Pid_4051\Launchpad_Open` and then `%LaunchpadProOpen1.DeviceDesc%= DDInstall,USB\Vid_1235&Pid_4051` but neither would work. I'm guessing i want that to go up to Pid_405f eventually to mirror the 16 types the other has. I also added a reference in Strings to LaunchpadProOpen1.DeviceDesc = "Launchpad Open". Tried uninstalling / forcing drivers in Device Manager and it doesn't want to load the Novation USB one.

That Vid / Pid thing matched up properly based on my working Launchkey Mini so maybe there is hope here.

Also you'll probably need to take ownership of the file if you're going to edit it, then restart pc out of paranoia before hoping windows will let you give it this driver when you force it again.

dvhdr, 2016-06-14:
Hi - great thinking, @SeanM93! Sadly though I've tried it and it's no help. Note that you wouldn't see it work as you'd have invalidated the signature on the driver package by modifying the INF.

I did get the bus analyser out though. And the problem is that something is going wrong after a GetConfigurationDescriptor, juuuust before there ought to be a SetConfiguration to bring the USB MIDI to life. I'll look a little further today now that I'm on the trail....

SeanM93, 2016-06-16:
Cheers :) that makes sense. Thanks for looking into it

davidhtims, 2016-08-24:
Hi, I purchased the Launchpad Pro, specifically to use [Subsequencely](https://github.com/jrcurtis/subsequencely) with a PC.
I would just like to know please, is this problem fixable at all or is there something basically wrong with the design of Novation's Launchpad Pro?

dvhdr, 2016-08-30:
It's fixable in theory, but I've not been able to find the problem. We moved the open firmware away from the proprietary toolchain we use for the factory firmware. The issue has been introduced during that change, and it's proving hard to find.

I can't make a commitment to fixing it at this stage, sorry.

dvhdr, 2017-03-14:
Thanks to some help from Os at Expert Sleepers, I think I've fixed this! I've merged the fix into the  `default` and `raw-adc` branches. The `flash-storage` branch is now stale.

dvhdr, 2017-03-14:
Sorry, forgot to close.

closed 2017-03-14 by dvhdr

mat1jaczyyy, 2018-08-18:
Any plans to implement the open-source firmware to the USB Driver? It would be nice to be able to use it with multiple applications, sometimes this allows for stuff that is normally not possible to do. I've tried implementing the ['workaround' by SeanM93](https://github.com/dvhdr/launchpad-pro/issues/9#issuecomment-225799784) and after disabling signature enforcement, the driver works as expected and allows me to control Launchpad Open ports from multiple applications, however this is not a full solution because it requires both messing around with the INF file and the OS startup parameters.