dvhdr/launchpad-pro #11: Vagrant / Toolchain setup

jaysonturner, 2015-11-07:
I had a little trouble setting up the build environment and just wanted to document it so that some poor soul in the future may benefit from these words.

I followed the instructions provided:

```
$ vagrant up
$ cd /vagrant
$ make
```
Then I got the following output:
```
vagrant@vagrant-box:/vagrant$ make
mkdir -p build/src/
arm-none-eabi-gcc -c -Os -g -Wall -I. -D_STM32F103RBT6_  -D_STM3x_  -D_STM32x_ -mthumb -mcpu=cortex-m3 -fsigned-char  -DSTM32F10X_MD -DUSE_STDPERIPH_DRIVER -DHSE_VALUE=6000000UL -DCMSIS -DUSE_GLOBAL_CONFIG   -g3 -ffunction-sections -std=c99  -mlittle-endian -Iinclude -I -o -MMD -o build/src/app.o src/app.c
make: arm-none-eabi-gcc: Command not found
```
Uh oh :(

So I initially thought I just need to install the ARM toolchain. So I did:
```
$ sudo apt-get install arm-none-eabi-gcc
```
This also caused problems, running make again delivered the following error:
```
vagrant@vagrant-box:/vagrant$ make
mkdir -p build/src/
arm-none-eabi-gcc -c -Os -g -Wall -I. -D_STM32F103RBT6_  -D_STM3x_  -D_STM32x_ -mthumb -mcpu=cortex-m3 -fsigned-char  -DSTM32F10X_MD -DUSE_STDPERIPH_DRIVER -DHSE_VALUE=6000000UL -DCMSIS -DUSE_GLOBAL_CONFIG   -g3 -ffunction-sections -std=c99  -mlittle-endian -Iinclude -I -o -MMD -o build/src/app.o src/app.c
arm-none-eabi-gcc -Tstm32_flash.ld -u _start -u _Minimum_Stack_Size  -mcpu=cortex-m3 -mthumb -specs=nano.specs -specs=nosys.specs -nostdlib -Wl,-static  -g3 -N -nostartfiles -Wl,--gc-sections -o build/launchpad_pro.elf build/src/app.o lib/launchpad_pro.a
arm-none-eabi-gcc: error: nano.specs: No such file or directory
```
A bit of googling lead me to this post: https://launchpad.net/~terry.guo/+archive/ubuntu/gcc-arm-embedded

It seems there's some oddness with Ubuntu 14.04 and the ARM toolchain installation. I had to change the repo that apt-get was using. So I did:
```
$ sudo apt-get remove binutils-arm-none-eabi gcc-arm-none-wabi
$ sudo add-apt-repository ppa:terry.guo/gcc-arm-embedded
$ sudo apt-get install gcc-arm-none-eabi
```
Now everything works fine and dandy :)

dvhdr, 2015-11-09:
That's really odd! At first I thought you'd not done `vagrant ssh` - but looking at your output it seems you did (you did, right?). It appears your vagrant box didn't install correctly, or more accurately it didn't complete the provisioning script which installs the build tools.

This kind of thing can happen if you log in to the box via the GUI before the script has finished, but it seems unlikely that you did that seeing as you're using the terminal. So I'm confused! Please can you tell me what platform you're on?

Glad you're up and running, and sorry you've had to go through the pain that the Vagrant system is intended to protect you from!

jaysonturner, 2015-11-10:
No need to apologise, oddness happens all the time.

I'm on El Capitan, I think I did login to the GUI first, before doing `vagrant ssh` so you may be correct. It seems that eclipse didn't install either.

Tell you what, tonight, I'll do a fresh clone and see what happens, just to be sure :)

dvhdr, 2015-11-10:
You might want to drop the GUI / Eclipse approach entirely, I know some folks find it unhelpful.  You can chop a bunch of downloads out of the provisioning script, and save some RAM / disk as well.

jaysonturner, 2015-11-10:
Coooooooool :)

faroit, 2016-05-16:
@jaysonturner this is now described in #17. Does this work for you?

JacobiusWrex, 2016-07-03:
I am stuck \

I downloaded vagrant and virtual box, rebooted, and when I type 'vagrant up' it shows me that error message about driver not being found. 

Quite confused! Any suggestions?

JacobiusWrex, 2016-07-03:
file:///home/ptpc/Desktop/Link%20to%20Downloads/launchpad%20vagrant%20up%20not%20up.PNG

drakmaniso, 2016-09-26:
Hi.

It seems I have run into the same problem, the "vagrant up" command doesn't complete correctly:

```
==> default: --2016-09-26 11:45:29--  https://launchpad.net/gcc-arm-embedded/4.8/4.8-2014-q3-update/+download/gcc-arm-none-eabi-4_8-2014q3-20140805-linux.tar.bz2
==> default: Resolving launchpad.net (launchpad.net)...
==> default: 91.189.89.223
==> default: ,
==> default: 91.189.89.222
==> default: Connecting to launchpad.net (launchpad.net)|91.189.89.223|:443...
==> default: connected.
==> default: HTTP request sent, awaiting response...
==> default: 502 Bad Gateway
==> default: 2016-09-26 11:46:29 ERROR 502: Bad Gateway.
==> default:
==> default: bzip2: Compressed file ends unexpectedly;

[... plus a few more lines of bzip2 and tar complaining]
```

Maybe there's a problem with the gcc-arm-embedded package url?

Note that I also had to change a line in the Vagrant file, the vm box "drmyersii/ubuntu-desktop-14.04-x64" isn't available anymore, has been replaced with "drm2/ubuntu-14.04-desktop-x64". (should I open a separate issue for this?)

dvhdr, 2017-01-10:
Cheers @drakmaniso - finally committed this!

closed 2017-01-10 by dvhdr