dvhdr/launchpad-pro #12: skip vagrant on OS X

faroit, 2015-11-13:
I really like the vagrant workflow since it increases reproducibility and cross platform portability. But... it's kind of slow and needs some effort to setup.

I just found out that compiling on OS X is actually very easy if the ARM eabi toolchain is installed. To make this even easier, I created a homebrew tap. That way OS X (homebrew installed) can start cross compiling the launchpad code simply by:

```
brew tap faroit/eabi48
brew install gcc-arm-none-eabi-48
make
```

Since it's that easy, I would propose to add this to the README and/or create an additional OS X bash script. I can compile a PR if you like

joefocusrite, 2015-11-13:
Just tried this out and it works a treat. I'd be in favour of this over the Vagrant workflow, too. Although they're not mutually exclusive.

jaysonturner, 2015-11-13:
<3

dvhdr, 2015-11-13:
Nice one @faroit - send me a PR and I'll merge it!

faroit, 2016-04-27:
added in #17

closed 2016-04-27 by faroit