dvhdr/launchpad-pro #13: Mini support ? (and other models too)

Patrik356b, 2015-11-29:


dvhdr, 2015-11-30:
Sorry, this won't work with Launchpad Mini or Launchpad S. Those devices always operate with a USB host, so you're better off implementing custom behaviour there instead. Thanks for your feedback!

closed 2015-11-30 by dvhdr

matiasmorant, 2023-04-14:
Could you expand on your comment above? I have a LP mini mk3. Would like to repurpose side buttons to change instruments

mat1jaczyyy, 2023-04-14:
> Could you expand on your comment above? I have a LP mini mk3. Would like to repurpose side buttons to change Vst instruments

this is better suited for a firmware assembly hack, but this project might prove interesting: https://github.com/gabonator/Projects/tree/master/NovationLaunchpad