dvhdr/launchpad-pro #15: Add Live API support?

jonleebrown, 2016-01-18:
Hi All, 

I wanted to add a vote for including the LiveAPI (including the JavaScript API) in the open source FW for the Launchpad Pro.

I've programmed Max for Live devices for other Launchpad models and found the API really useful--but it seems like the Pro supports a different and, in some ways, feature-reduced interface. 

I can't observe button presses or light buttons via the send_value functions (or at least I am clueless about how to fully explore and understand the LP Pro interface).

I would be willing to implement this myself if it were made available in the FW here.

Thanks,

Jonathan

mluis, 2017-06-24:
I just opened an issue related to real-time streaming, is this Live API you're talking about related to it somehow? I'm very interested on the subject.

dvhdr, 2018-11-08:
Check out the amazing https://github.com/mat1jaczyyy/lpp-performance-cfw by @mat1jaczyyy

closed 2018-11-08 by dvhdr