dvhdr/launchpad-pro #25: Return of the Dirty Vagrant, Part III

oasis9, 2017-05-15:
It seems that using the vagrant box from drmyersii , now drm2, has certain repercussions. Either this is not the box belonging to the original author, or it is, and eclipse on said box is broken (which it is). Java needs to be updated to Java 8 and the eclipse release (broken - icon missing, won't start) needs to be removed and the C/C++ edition installed. After completing this, I can finally work on an app of my own. I'm overjoyed, even though a bit of debugging and witch-hunting (I kid) was involved.

oasis9, 2017-05-15:
This isn't a very well structured issue. The solutions I provide are patches (could temporarily be included in readme). I am unsure how to produce a vagrant box that would solve this issue, so at this time I haven't. I would do this if I knew how, but that is for another day. @dvhdr, you could either wait until then to update the link in your vagrantfile, or accept my pull request, update your readme, and complete that later. Thanks for the work you've done on this - It's beyond me.

granadaxronos, 2017-06-08:
Hello.... I'm about to setup the environment now. Your tips will help me a bit.
I'm downloading the drm2 box now.

oasis9, 2017-06-08:
If you have any issues, message me :3

oasis9, 2017-06-08:
I was going to post a tl;dr, but I forgot how I had it set up as I am away from my computer for the next few hours.

granadaxronos, 2017-06-09:
I think the installation of vagrant and other UBUNTU OS stuffs are redundant for the compiling of the codes. I am installing a Windows X64 eclipse. I will let you know of the test result.

dvhdr, 2017-06-09:
@granadaxronos - you don't even need Eclipse, all you need is the gnu-arm-eabi tools.

oasis9, 2017-06-09:
I wouldn't agree with the latter comment of @granadaxronos; this repository, even in the anomalies of its approach, is very useful. I have had a very non-traditional introduction to development as a whole and am excited to possibly begin attending University next year, which will allow me to grow my theoretical knowledge. Besides that, for someone such as myself with a lack of knowledge in terminology, the current ease of setup for this project is wonderful (at least I can say that now that I've figured it out).

dvhdr, 2017-10-24:
I dunno @oasis9 , he's right in the sense that I am indeed a complete nut :)

closed 2017-10-24 by dvhdr

oasis9, 2017-10-30:
naww
no need for that 😋 this was definitely fun while it lasted, though I've become so tied up in other work that I don't have time to get it all set back up after reinstalling Windows on my shiny new SSD and not to mention that bastard 2 minute upload 😢

dvhdr, 2017-10-30:
Yeah the two minute upload does make debugging pretty horrible. Did you try the simulator on macOS? I think the creator of Subsequencely made a really nice desktop one.