dvhdr/launchpad-pro #26: Enabling real-time streaming to remote service endpoint.

mluis, 2017-06-24:
Hello all,

**TL;DR:** I'm asking for a Yes or No theoretical answer to the question: Is it possible to have the Launchpad to stream in real-time to a remote service endpoint?

**OK** Let me Read: 
I just purchased my first brand new Launchpad (mk II) 3 days ago and I am very thrilled with the capabilities of such device. As if that would not be enough I was completely surprised when I found, after that, that actually the firmware was open source and this really triggered me touching my heart and emotions. So that being said my question is if enabling real-time streaming to a remote service endpoint:

1. Is it theoretically possible? (asking just to save me some time on research)
2. Is it being done already?
3. Is it something the launchpad community would find purpose on using it?
4. Is there any place more suited to this discussion?

I have a lot more to ask and discuss. It all depends on the 1st "spark".

Thanks

mluis, 2017-06-24:
Ok, I just looked at what you're making available. The public API is quite short and I have a doubt also, the 128k of flash is what is available for the apps, libraries included?

dvhdr, 2017-06-26:
Hey @mluis ! Just to be clear, this repo is for the Launchpad Pro, and will not work with Launchpad Mk2. The Pro has round buttons on each edge, but the Mk2 only has them on the top and on the right.

Hope that makes sense! I'll try to answer your other questions if they're still relevant :)

mluis, 2017-06-26:
Hi @dvhdr I thought the answer would touch that subject, sorry if I didn't decoupled the subjects. That just would mean I would have to buy a Pro. I'm still interested in the answers though! From my experience in music buying a Pro doesn't justify the investment but if I can code for it that would put a cherry on the top of the cake. =)

mluis, 2017-07-06:
@dvhdr just waiting for some input of you :)

dvhdr, 2017-07-18:
So... I'm struggling to understand what you mean by realtime streaming. I'm guessing you want to send MIDI messages over a network somehow? In which case, this firmware isn't really what you need. You'd be better off creating a desktop app (or finding one that already exists, perhaps an OSC app) that translates USB MIDI from the Launchpad and sends it over the network. Perhaps as OSC packets, or as RTP-MIDI messages.

Is that the kind of thing you're after?

Thanks for the Docker PR, it's nice!

mluis, 2017-07-18:
I have time. :) My intention is not building a desktop app. I would rather use something like IP-over-USB running on a kernel's host machine ( *somehow* ;) ). I can get a TCP/IP stack in much less than 128Kb of memory, so I want to embed that there and that's not the problem. I also can do the math running on the LPPro the pointers over the mem pages but I would rather like to know how much space is left on the device for a compiled 'Hello World' app since you may have access to a device much faster :P.

Let's talk! :)

dvhdr, 2017-07-19:
I see (I think :). On the device you have about 90K of flash left over for your app code. The really scarce resource is RAM - you have only 20K, of which the runtime uses about 9K (IIRC). So not a lot of room to play with! If you're getting close, there are a few things I can do to free up a bit of that RAM, at the expense of some features of the API.

I don't know anything about IP over USB, so I can't help you there.

mluis, 2017-07-19:
Saving space is always important but 11K of RAM is a lot more than I was expecting (2KB) so there's plenty of room right now. :)

For now, I will try to setup an ARM Cortex M3 emulator using qemu and flash it with the compiled code of the LPPro. I'll keep you posted.

IP over USB has nothing special, in Linux it's supported natively, OSX and Windows I'm not updated yet but it's just less than half a dozen of command line commands or setting a config in /etc/network/interfaces

dvhdr, 2018-11-13:
Closing this one as I don't think there's anything else I can add.

closed 2018-11-13 by dvhdr