dvhdr/launchpad-pro #33: Provide means for declaring multiple MIDI ports over USB

ab5tract, 2017-11-14:
I would like to write a firmware that provides a USB MIDI port per function. This would allow the note mode to be routed to one instrument, the CC faders to another, and a sequencer and user mode to two more.

I was a bit disappointed to find out that the number of ports over USB is currently hardwired.

dvhdr, 2017-11-20:
Hmmm, interesting idea. The would require changes to the USB descriptors and low level firmware, which is not currently part of this release. We'll give it some thought, sorry you couldn't achieve what you wanted.

ab5tract, 2017-11-20:
Thanks for looking at the ticket.

My current workaround plan is to (re)create the base functionality of the
shipped firmware, but with different MIDI channels per mode.

This would allow the creation of virtual MIDI devices that send/receive
from those specific ports.

dvhdr, 2018-11-08:
Dynamic port changes are out of scope for now, sorry. For some awesome work on recreating factory firmware features, check out https://github.com/mat1jaczyyy/lpp-performance-cfw

closed 2018-11-08 by dvhdr