dvhdr/launchpad-pro #37: Installing Development Environment on Windows 10 Using Vagrant

scottboyan, 2018-04-21:
[image]

I am trying to configure a development environment using vagrant.  I cloned the project to my local machine.  Installed VirtualBox. Installed Vagrant. Navigated to the project directory on my local machine (it has a Vagrant file in it) and used the command "vagrant up." The attached screenshot shows the message that I'm getting.  You can see at the very top I'm in the right directory, because you can see the vagrant file in it. I also tried "vagrant init" and I got a message that I could not, because there is already a vagrant file there. So Vagrant is seeing the file. 

Do you have any ideas for what I should try? Is the vagrant file from an older version that maybe the one I installed today can't use? Any suggestions?

dvhdr, 2018-09-07:
Ick, that's odd. I don't have a Windows Vargrant environment that I can try to reproduce this with. A quick fix would be to use the Docker version, or install the ARM tools natively.

dvhdr, 2018-09-07:
Oh my, just saw the date on this post. Sorry for the delay!