dvhdr/launchpad-pro #38: Access Bootloader Text toggle

mat1jaczyyy, 2018-08-18:
In the bootloader, there is a configurable toggle for Text scrolling upon mode selection on stock firmware. Is there any way to get or set its value from the open-source firmware?

dvhdr, 2018-09-07:
No, sorry - the bootloader is completely separate from the open firmware and can't be modified.

mat1jaczyyy, 2018-09-07:
Not asking for modifying the bootloader, just a means to access the value of the one configuration boolean on the right side in the bootloader.

dvhdr, 2018-09-07:
Oh, sorry, reading comprehension fail!

In principle this is doable, it's just an area of flash. In practice, we chose not to expose writes to the whole of flash memory, to avoid overwriting bits of program, other settings etc.

We could expose this as a feature, so I'll tag this issue as an improvement, but it's unlikely that we'll get to it any time soon. What's your use case?

mat1jaczyyy, 2018-09-07:
In particular, I'm recreating the stock firmware's functionality in full, it would be a nice touch to get the scrolling text toggle back. But, it could be used for countless other things, for example enabling debug logging over SysEx.

dvhdr, 2018-11-14:
Implemented, along with an option to read the bootloader USB device ID.

closed 2018-11-14 by dvhdr