dvhdr/launchpad-pro #47: LED flicker

mat1jaczyyy, 2018-11-21:
I've noticed all LEDs flicker significantly, even if they aren't plotted by app_timer_event. It's not enough to capture on my camera, but definitely visible to the naked eye. The LED states are not changing, I plot them once and don't re-draw anything while it's flickering.

This was not the case prior to merging #46, maybe it has to do with something that was changed when hal_read_led was implemented? I'm not using that HAL call at all in my code.

dvhdr, 2018-11-22:
This is really odd - I can't reproduce (LEDs are rock solid for me). Can you send me a code snippet?

It's possible that the additional time in ADC scanning is holding off the LEDs in some cases. Might be processor dependent - is it a minimal app or is there lots going on? Do you see the flicker by, for example, modifying the example app to display LEDs (and of course removing the ADC display code)?

mat1jaczyyy, 2018-11-24:
I can't get it to happen by modding the example app at all. I do notice that dim colors (max value 15ish) are easier to notice. 

There's a fair amount of stuff going on, and it happens in my custom firmware project in all use cases I've been able to check. You can easily notice the flicker on my Setup screen, where the dim LEDs can definitely be seen flickering.

I've pushed your upstream changes to the [api-improvements branch in mat1jaczyyy/lpp-performance-cfw](/mat1jaczyyy/lpp-performance-cfw/tree/api-improvements), should you want to take a look.

mat1jaczyyy, 2018-11-24:
I've isolated the issue to [https://github.com/mat1jaczyyy/lpp-performance-cfw/blob/api-improvements/src/other/tempo.c#L36](my tempo keeping mechanism) for Channel 2 / Channel 3 recreation. If the call to this function in [https://github.com/mat1jaczyyy/lpp-performance-cfw/blob/api-improvements/src/app.c#L44](app_timer_event) is removed, the issue disappears.

mat1jaczyyy, 2018-11-24:
I've just pushed a commit to the branch mentioned above that splits up the 2 for loops that draw flash/pulse into 4 tick calls / frames. This gets rid of the issue, so it's likely that the hal_plot_led function has gotten more expensive to call because of the internally managed buffer for hal_read_led returns.

Please check out my findings and let me know if there's anything we can do about it.