dvhdr/launchpad-pro #53: Open source launchpad.c to support C++ compilation

Kyran-C, 2019-07-12:
From my understanding, in order to support C++, `main` needs to be compiled by the C++ compiler. As it stands now, that's precompiled into `launchpad.a` and linked during build. Would it be possible to refactor things to have `launchpad.c` compiled by us and maybe have it link to a `launchpad_lib.a` which would have the opaque guts you don't want open sourced? If there's proprietary stuff you don't want public, that could be put into the lib. Alternately, if it's just hidden for simplicity's sake it might be worth exposing so we can do some deeper tinkering.

Either way, it'd be really amazing to be able to use templates and member functions and lambdas and bools and function overloads and...

mat1jaczyyy, 2019-07-12:
If you open it with a static analyzer (I use ghidra), the binary appears to have all symbols and function names inside (except for local variables) and is modifiable / probably decompilable. Given how slow the responses have been over time, this is probably the better, but tougher bet.

tgroh007, 2019-07-12:
Hey guys, this is pretty cool, I gave up on my mpk49 pads and purchased the lpp a few months ago and take it everywhere along with my mac book. @Kyran-C I was reading along in the enhancements comments and have modded the crap out of my mpk49 as well and pretty much haven't touched it since getting the lpp. Anyways I will be along for the ride, playing around with the firmware as well. Chat with ya all soon!!

tgroh007, 2019-07-16:
Didn't have any luck setting up the docker container environment using the info from this github, but this repo worked fine https://hub.docker.com/r/mluis/launchpad-pro-dev. Gonna mod my llp's pads soon then start testing. Thanks all for paving the way!!

zeroeth, 2020-01-19:
I haven't tested it with the launchpad pro, but you can use extern "C" to as a bridge between the c and c++ code, as noted https://isocpp.org/wiki/faq/mixing-c-and-cpp#call-cpp

mat1jaczyyy, 2020-01-19:
From what I understand this dude took my firmware and added C++ support to it, as well as his own tweaks. https://github.com/DarioDaF/lpp-performance-cfw