dvhdr/launchpad-pro #55: Launchpad Pro Mk3 support

faroit, 2020-01-30:
We all appreciated the open source move for the LPP. Now that the new version is out devs and pro users will likely be interested in 

* will it support the open firmware?
* will the sensitivity issues #6 be addressed?

mat1jaczyyy, 2020-01-30:
Also curious about open firmware for Pro MK3.

raymak, 2020-03-12:
Interested in open firmware for Pro MK3 as well

mat1jaczyyy, 2020-05-03:
Either way, the checksum's been reversed, it's just regular crc32. I had used a reimplementation of it to hack together some X and Mini MK3 firmware that featured a proper Drum Rack layout with side buttons included before Legacy mode was released.

Pro MK3 seems to be running on multiple processes/threads, it's using FreeRTOS. Funnily enough debug asserts are compiled in.

henning, 2020-11-30:
@dvhdr as this is open since nearly a year soon - could we get an answer on this, please? Would really be great to know if this is a possibility! Simple quick yes/no would be really nice...

BTW, even though slightly off topic: it would also be really really great, if the sources of the components JS app would be published, there are tons of features to be added and some bugs to be fixed... (e.g. it fails - as in "does nothing when communicating with the device" - on Linux even though connection is "green", should be supporting not only chrome but also chromium, the circuit editor should be able to select multiple samples to import at once, and a bunch of other additions that your users would love to add, so you can concentrate on making great machines while we add to the editors what we need)

luzhongdao, 2020-12-14:
I contacted Novation Technical Support asking this question and linked this thread. Sounds like a no:

> Thanks for your email.
>
> Take a look at r_cycles, our open source solution for hacking our Launchpads: https://novationmusic.com/en/news/hack-making-music
> 
> This is our solution to open source firmware for these devices.

luzhongdao, 2020-12-14:
I contacted Novation Technical Support asking this question, and linked this thread.

> Thanks for your email.
> 
> Take a look at r_cycles, our open source solution for hacking our Launchpads: https://novationmusic.com/en/news/hack-making-music
> 
> This is our solution to open source firmware for these devices.