dvhdr/launchpad-pro #61: Unclear how to send a program change message

fuzzballb, 2023-05-17:
To make a clip launcher (for the MC-707), i need to know how to send a program change, unfortunately I can't find example code on how to do this. 

Is there a comprehensive guide that covers the basics?

Thanks in advance

mat1jaczyyy, 2023-05-17:
hal_send_midi should work
use the standard byte id for program change from midi specification (i think its 0xCz where z is the channel you want, but it's off the top of my head i might be wrong)

fuzzballb, 2023-05-18:
Tried these and looking at the MIDI monitor, it seems to work correctly.

// MIDI CHANNEL 3 PROGAM 67
hal_send_midi(USBMIDI, 0xC2, 0xC3, 0);

For the DIN midi out i will need to use DINMIDI

Seeing how the MIDI channel is part of the STATUS (0xC2), I can see why it isn't part of the status constants.

closed 2023-05-18 by fuzzballb